romainbrette/holypipette #25: Hang-ups with Qt livefeed

romainbrette, 2017-10-20:
The camera window still hangs up some times, without any error displayed. The whole Qt thing seems to be frozen (clicks don't trigger events any more).

romainbrette, 2017-10-20:
Actually it's just the livefeed component, because clicks don't work but keys do

romainbrette, 2017-10-20:
This has to do with the camera, because the camera doesn't provide correct images anymore

romainbrette, 2017-10-20:
Actually it might be that clicks work (not sure), so the problem would be with the camera

mstimberg, 2017-10-20:
Ok, that makes sense, so it is most likely still an issue with the micromanager camera code (unfortunately the source code for the Lumenera interface is not open...). I'll have a another look if I find anything in Micromanager itself, otherwise probably the best workaround would be if only the livefeed regularly asked the camera for a new image, and all other code (calibration, etc.) just gets the most recent image. This should isolate us more from the internal micromanager code, issues with threads, etc.
But then, ACQ4 seems to use the same approach as we do (I'll recheck), so it might still be a bug in our code after all...

romainbrette, 2017-10-20:
I added a check for new_image() in livefeed; let's see if it makes a difference.

romainbrette, 2017-10-20:
nope

romainbrette, 2017-10-20:
I can see that sometimes the camera seems to freeze for a fraction of second. Maybe related.

mstimberg, 2017-10-20:
I had a look at how ACQ4 uses the camera, it is quite different: there's a separate "acquire thread" that continuously empties the buffer and then emits "new frame" signals for each frame. It does not use `getLastImage`, but only `popNextImage`. However, calling `getLastImage` without ever clearing the buffer does not seem to be wrong, it is used for example in one of the official MicroManager examples: https://github.com/radioxoma/micromanager-samples/blob/master/mm_live_video_controls.py

I'll try whether I can reproduce the issue on my machine using MicroManager's "Demo camera" -- of course if it it only triggers with the Lumenera cam, we are out of luck...

romainbrette, 2017-10-23:
No more hangups today but sometimes the program exits with code 1

romainbrette, 2017-10-26:
The fix doesn't work!
   Camera image is *exactly* the same as 1s ago -- resetting camera
but then the image is still frozen on screen...

romainbrette, 2017-10-26:
Worse: when I relaunch, the camera still doesn't work...

romainbrette, 2017-10-26:
Then I launched umanager and it worked...

romainbrette, 2017-10-26:
Alternatively if I relaunch the python script several times it works again.

romainbrette, 2017-10-26:
Actually camera reset works.

romainbrette, 2017-10-26:
Now I get no image available, waiting for 5ms
repeated ad lib

romainbrette, 2017-10-26:
The only significant difference I see with Aurélien's code is he was running camera acquisition continuously in a thread (as opposed to every 50 ms)

romainbrette, 2017-10-27:
Today it seems to work fine... so just the cable?!!

mstimberg, 2017-10-27:
Frustrating and reassuring at the same time 😉