romainbrette/holypipette #51: Bug in camera.py with tracking

romainbrette, 2018-07-26:
Traceback (most recent call last):
  File "/Users/Romain/PyCharmProjects/holypipette/patch_gui_with_tracking.py", line 20, in <module>
    gui = TrackingPatchGui(camera, pipette_controller, patch_controller)
  File "/Users/Romain/PyCharmProjects/holypipette/holypipette/gui/patch.py", line 62, in __init__
    with_tracking=True)
  File "/Users/Romain/PyCharmProjects/holypipette/holypipette/gui/patch.py", line 20, in __init__
    with_tracking=with_tracking)
  File "/Users/Romain/PyCharmProjects/holypipette/holypipette/gui/manipulator.py", line 18, in __init__
    super(ManipulatorGui, self).__init__(camera, with_tracking=with_tracking)
  File "/Users/Romain/PyCharmProjects/holypipette/holypipette/gui/camera.py", line 362, in __init__
    with_tracking=with_tracking)
  File "/Users/Romain/PyCharmProjects/holypipette/holypipette/interface/camera.py", line 19, in __init__
    self.multitracker = cv2.MultiTracker_create()
AttributeError: 'module' object has no attribute 'MultiTracker_create'

It seems to be an opencv version problem.

romainbrette, 2018-07-26:
Perhaps this only exists in OpenCV 3? (I use 2.4.11)

mstimberg, 2018-07-26:
You need OpenCV 3.4 -- but OpenCV 3.4 for OS X is in the standard anaconda distribution, so a `conda update opencv` should be enough.

romainbrette, 2018-07-26:
It works, thanks.

closed 2018-07-26 by romainbrette