romainbrette/holypipette #60: Automatic range measurement

romainbrette, 2018-07-27:
For calibration and also movements, we need to know the minimum and maximum positions of the axes. Several ideas:

1. One way is to ask the user to move all axes to one endpoint, then to the other endpoint. This would mean two more keys (could be just one key: "end point").
2. Another is to continuously measure the position of all axes (or say every second), and store the min and max over the entire session.
3. A hybrid strategy is to have one key to record/stop the range of axes, where the user moves the axes in the reachable region and measure the min and max.

romainbrette, 2018-07-27:
The coverslip (floor) could also be measured in this way. But then one has to be careful not to focus beyond the coverslip. This might rule out (2). I would favor (3).

mstimberg, 2018-07-27:
From my side, I'd also prefer (3). Maybe with continuous visual indication about the current axis ranges to avoid forgetting to move one of the axes.

For the coverslip, we don't need a range but only a single value does it make sense to link it to the range measurement?

romainbrette, 2018-07-30:
Ok I've implemented (3). It also updates coverslip floor (=microscope.min). I'm also not entirely sure about where in GUI or interface those things should go (it's spread over the two now). Finally I'd like to display manipulator ranges, but these are 18 numbers... What I do here is that a debug line is displayed if any of the axes has a range < 500 um, at the end of the measurement procedure.

The procedure is currently as follows:
* Press M starts the measurement process. All ranges are reset to 0.
* Every 500 ms, all axes are measured. Min and max values are updated.
* Press M stop the measurement process.
* If any of the ranges is smaller than 500 um, a debug line is displayed.

romainbrette, 2018-07-31:
I just tested it on the rig. It works, but a few issues/remarks:
* It would be indeed useful to display the current ranges.
* For the stage and microscope, this system is actually not great. For the microscope, the focus can be moved beyond the coverslip but we don't to go beyond the coverslip. For the stage, we can move it to places where there is no light anymore. When adjusting the ranges, we therefore have to be careful not go to those unwanted locations; otherwise, we have to redo the whole procedure again.
* For the microscope, ranges for the calibration might be different from ranges for use. However, we may assume that those are larger in any case, so maybe it's fine.

For the second point, one could use proposition (1). Alternatively, the user hits a key every time he considers the current position to be acceptable. Then the program takes the min and and max of all measurements at those times. In other words, same as (2) except measurement is only done at the time of a key press. But then it would seem necessary to be able to reset the ranges (another key).

romainbrette, 2018-07-31:
Alternative idea: **automate range measurement**:
* The user removes pipettes from the manipulators.
* The microscope Z is moved down and stops at the coverslip with an autofocus algorithm.
* The stage is moved while global luminance is measured. Ranges are calculated so that luminance is above some threshold relative to the maximum (say 90%).
* Each manipulator axis is moved to measure min and max positions.

Could we put the necessary stuff in the FakeCamera? That is:
* Coverslip is simulated by having a spatial pattern (e.g. some periodic grating) convolved with a Gaussian whose variance increases linearly with Z-Z0 (Z0 = coverslip).
* Light boundaries simulated as a mask (black outside of disk).

romainbrette, 2018-09-11:
One problem I noticed is that motor ranges are automatically loaded at the beginning, so the default ranges (-infinity to infinity) are not useful.