romainbrette/holypipette #79: Bug in calibration

romainbrette, 2018-11-05:
I got this while calibrating:

Traceback (most recent call last):
  File "C:\Users\Romain\PycharmProjects\holypipette\holypipette\gui\livefeed.py", line 91, in update_image
    self.display_edit(scaled_pixmap)
  File "C:\Users\Romain\PycharmProjects\holypipette\holypipette\gui\camera.py", line 431, in display_edit
    func(pixmap)
  File "C:\Users\Romain\PycharmProjects\holypipette\holypipette\gui\manipulator.py", line 98, in draw_scale_bar
    self.interface.microscope.position())
  File "C:\Users\Romain\PycharmProjects\holypipette\holypipette\devices\manipulator\microscope.py", line 43, in position
    return self.dev.position(self.axis)
  File "C:\Users\Romain\PycharmProjects\holypipette\holypipette\devices\manipulator\luigsneumann_SM10.py", line 154, in position
    res = self.send_command('0101', [axis], 4)
  File "C:\Users\Romain\PycharmProjects\holypipette\holypipette\devices\manipulator\luigsneumann_SM10.py", line 137, in send_command
    raise serial.SerialException(msg)
SerialException: Expected answer '060101', got '' instead

Maybe it's an interference between the calibration process and the camera display process?

We're at the rig now.

mstimberg, 2018-11-05:
There's a lock in place that prevents new commands from being sent while we are waiting for the response from another. I just made another small correction, though, but it should only matter for commands which expect a special kind of response, I don't think we are using any of those... Still, can you check whether it helps? Otherwise I'm going to add debug statements.

romainbrette, 2018-11-09:
It seems to be working now.

closed 2018-11-09 by romainbrette