romainbrette/holypipette #80: Bug in saving image

romainbrette, 2018-11-12:
We got this:
DEBUG:PIL.Image:Error closing: 'Image' object has no attribute 'fp'

mstimberg, 2018-11-12:
Did it actually fail to save the image and/or crash? E.g. this link seems to say that this message is harmless: https://stackoverflow.com/questions/48338268/imageio-or-celery-error-closing-image-object-has-no-attribute-fp

romainbrette, 2018-11-12:
It didn't crash but failed to save the image.

mstimberg, 2018-11-12:
I  pushed a change that follows the suggestion in the above stackoverflow link to use the PIL module directory. Let me know whether this works (everything works fine on my machine either way, so I cannot really test it).

romainbrette, 2018-11-15:
Yes it works. In fact we realized it actually worked despite the error, just that for some reason there is a "desktop" both on C: (fake?) and D: (real one).

closed 2018-11-15 by romainbrette