romainbrette/holypipette #82: Paramecium tracking

romainbrette, 2018-12-10:
It's still not very good. I have a good video here:
http://82.253.119.171/owncloud/index.php/s/Uvq6N0otsUevWCV
and I use 0.62 pixel/um for the setup_script.

One thing that might be useful is to be able to display processed images (eg contours) (to replace the video), I guess that should be simple enough? Marcel could you perhaps add a couple of lines just to show me how to do this? (I guess in where_is_paramecium and the caller script ?)

mstimberg, 2018-12-14:
I did a few changes to the code and refactored the tracking a bit. More of the logic is now included in the `paramecium_tracking.py` file (the idea being that this could be useful independent of the GUI). The `where_is_paramecium` function also moved into a `ParameciumTracker` class (that stores previous positions) and has been renamed to `locate`.
This `locate` function now returns an `info` dictionary together with the coordinates/size of the best fitted ellipse. This dictionary contains stuff like all the detected contours and the fitted ellipses. I added two options to the paramecium configuration (under "debugging") that switch on drawing these contours and/or ellipses. It's probably too much, but it should give an idea how to use this information. If everything is switched on it plots:
* contours that were too short or did not have at least 5 points (**yellow**)
* contours that were good enough to be fitted by an ellipse (**red**)
* contour that was used for the best-fitting ellipse (**green**)
* fitted ellipses that were not considered (too long/short, etc.) (**light blue**)
* fitted ellipses that were considered for the final choice (**purple**)

I also added a `slowdown` option to the `RecordedVideoCamera` class so that it shows frames from a video repeatedly (this is not implemented in any kind of clever way, e.g. the processing still happens for each frame, even if it did not change).

I made all these changes in the `refactor_tracking` branch because I did not test it very thoroughly yet and I did not want to break the master branch. Feel free to merge it into master if it works for you.

romainbrette, 2018-12-24:
Ah thanks, sorry I hadn't seen the notification! Looks nice.

closed 2019-01-03 by romainbrette