romainbrette/holypipette #95: Simplification of GUI/Interface system

romainbrette, 2021-06-08:
Currently, to set up a GUI for a particular application, we need to create three objects: a GUI, an interface and a controller. Each class is created in a separate folder (gui, interface, controller). It's complicated. I often find it unclear where the code should go (in which of these 3 objects). Also, for a question of organization, it would also be simpler if each application had its modules in the same folder.

romainbrette, 2021-06-09:
Basically, it would be nice if it were a bit simpler to create a new application. Currently, I think it's very hard for someone outside the project to make their own GUI.

mstimberg, 2021-06-10:
I'm sure there are things to simplify, but do you have a concrete example of things that are difficult at the moment? The architecture is the classical [MVC](https://en.wikipedia.org/wiki/Model%E2%80%93view%E2%80%93controller) architecture and I introduced it to make it possible to 1) write the low-level control code without having to refer to Qt, so that it can be used in contexts without GUI (scripts, tests, etc.) and 2) to automatically deal with all the nice things that Qt, threads, etc. can offer us, in particular the option to stop a long-running procedure. We have some quite extensive developer documentation on this topic that should hopefully make clear where things go: https://holypipette.readthedocs.io/en/latest/developer.html
Again, happy to change things but it would be good to have a concrete example and see where it is difficult to implement with the currnt system.