romainbrette/holypipette #96: Automatic Paramecium impalement

romainbrette, 2021-06-10:
The key element in automating Paramecium experimentation is the impalement, which uses an electrical signal. One option would be to import clampy and then access the board. However, this means that we cannot run anything else using the board simultaneously, eg an oscilloscope.

Alternatively, we communicate with an external program. For example, the oscilloscope program may send pulses regularly, and write the measurements to a file. This seems like the simplest option for semi-automatic operation.

An alternative would be that holypipette uses clampy to measure Vm, eg sending pulses, and writes to a file. Then the external program (oscilloscope) reads from the file. This would be best for fully automatic operation.

romainbrette, 2021-06-11:
For semi-automatic operation, I suggest the following:
- The user calibrates the two pipettes, and then the pipettes are withdrawn.
- The microscope goes to the center.
- The user focuses on the cell. The z position is recorded.
- The user clicks on the two target points for impalement. Photos are taken around those two points.
- The pipettes are moved in turn to each of those two points. Template matching is used to refine target position.
- In case of failure, the pipette is partially withdrawn then the target position is located again and another trial begins (assume Re hasn't changed).

As a start, we could add to the camera GUI a tracking functionality: we click on two (n) points, then the display shows the tracked points in real time.

The next step would involve automatic localization of cells. This could be done with:
1. deep learning (DLC)
2. background subtraction, requiring control of the pump
3. calculation of bounded variation (cilia motion)

I note that only option (1) would work with deciliated cells.

romainbrette, 2021-06-18:
Done (not tested with cell): impalement that stops when Vm drops.

To automate the pump, the simplest is to use the board and contacts, see the manual of Gilson Minipump 3 (https://fr.gilson.com/pub/media/docs/MINIPULS3_UG_LT801121-18.pdf). Alternatively, usb + arduino + contacts.

huyhoang93, 2021-06-21:
> Done (not tested with cell): impalement that stops when Vm drops.
> 
> To automate the pump, the simplest is to use the board and contacts, see the manual of Gilson Minipump 3 (https://fr.gilson.com/pub/media/docs/MINIPULS3_UG_LT801121-18.pdf). Alternatively, usb + arduino + contacts.

For the pump control, last time, I tried with arduino and we could control this one already. (holypipette/development/pump-control.py)

romainbrette, 2021-06-21:
Was it the same model?

huyhoang93, 2021-06-21:
Yes, it is the same model. the minipul 3 pump that Sarah used before.